nukeem/continuousprint #1: Default wipe script suggestions

KlaymoreGhost, 2020-04-25:
The default wipe script you have configured for the print waits for 26 minutes, then uses the GCODE to wait until the bed reaches 25C before continuing. I suggest swapping the order those are run in the GCODE. I.E. First set your temperature then dwell for a bit for stabilization. 

I noticed in testing that the hotend takes the shortest path to X110 Y235at the present height. and the entire Gcode runs first. So if your slicer settings are (nonstandard?) or if the print has multiple models and finishes on the foremost model, it could potentially run into the aftmost model. I suggest raising the hotend fully, then moving to the wipe position. 

There is another issue I've found. Several of the Gcodes I tried (stock settings from the slicers) disable the steppers as soon as the print job is finished. So I would Suggest enabling the steppers first thing.

I also changed the G1 to G0 (extruding vs non-extruding)

What I am currently testing:
;set absolute positioning
G90
;enable steppers
M17
;move hotend to 100 (could this be read to add 5 or 10 from the currently loaded Gcode? 100 is 5 above my test object)
G0 Z100
;set the bed temp to 30 degrees and dont do what next until it gets there
M190 R30
;Wait 90 seconds for temp stabilization
G4 S90
;verify temp below threshold
M190 R30
;move to back corner
G0 X200 Y235
;move to mid bed aft
G0 X110 Y235
;come down to 1MM from bed
G0 Z1
;wipe forward
G0 Y0
;wipe aft
G0 Y235
;autohome
G28

KlaymoreGhost, 2020-04-25:
All in all this is a great start to something people are trying to figure out manually adding gcodes together

nwpulver, 2020-04-25:
Maybe a script that generates the wiping gcode based off the object and printer would be useful!

nukeem, 2020-04-26:
Thanks really great suggestion, I'll incorporate some of those suggestions. I found that the M190 wait would end prematurely in my tests so added the wait so the bed would cool down enough. Ill do some tests with your suggestions and update the code.

nwpulver Good suggestion but unfortunately out of my ability currently. I'll keep it in mind though

KlaymoreGhost, 2020-04-26:
Paul,

I tested further and I was testing with a thin print base and just coming
back to see if it had worked after a few hours. On further review, I think
the dwell is necessary. At least in my stock Ender 3, it seems the printer
isn't honoring the cooldown command after all. It begins cooldown but then
moves on before the cooldown is complete.

nukeem, 2020-04-27:
Yep, this was occurring on my Ender 3 as well. I found in most cases on glass bed the bed needed to be completely cold for the print to pop off

grahamgraham, 2020-04-29:
My $0.02: 
I think it would be best to set the absolute positioning after you’ve repositioned the Z axis, only because then you don’t have to edit the tile for different objects. 

> M17
> G0 Z10
> G90

I also found that with my printer (Ender 3 with borosilicate glass bed) it needs to be completely cooled off, like what @nukeem mentioned. I set my temps to 26°, with a wait time of 20min.  I had an issue where a print stuck, even then, so I adjusted my speed of the wipe so it’s torquier. I used 

> G0 F500 Y0
> G0 F500 Y200

And it moves quick enough so it’s not awful and slow enough to pop anything off. 

Now if I can figure out how to remove the skirt and initial priming while away, I’d be golden.

rjelbert, 2020-05-05:
This is my v7 N3DPS face shield bed clearing script (Lulzbot Taz 6 + Aerostruder ONLY):

; bed clearing script for the v7 N3DPS face shield band
; placed -90 degree rotation and 50mm forward on a lulzbot taz 6 bed with Cura
; works ONLY with an Lulzbot Taz 6 with an Aerostruder

G28 ; home
M17 ;enable steppers
M91 ; Set relative for lift
G0 Z10 ; lift z by 10
G90 ;back to absolute positioning

G0 X140 Y280 ;move to middle  bed rear
G0 Z2v ;come down to 2MM from bed

; bump the right hand side by 5mm 
G0 X280
G0 Y145
G0 Y140 Z5 F1000
G0 Y220 
G0 Z2v

; move to middle bump point
G0 X220 Y189
G0 Y186 Z5 F1000 ; 3 mm
G0 Y220
G0  Z2v
; do again
G0 X220 Y189
G0 Y184 Z6 F1000 ; 5 mm
G0 Y220 
G0 Z2v

; move to left bump position and bump and lift
; 2mm bump and lift
G0 X160 Y197
G0 Y195 Z4 F1000
G0 Y220
G0 Z2v
; 3mm bump and lift
G0 X160 Y197
G0 Y194 Z5 F1000
G0 Y220
G0 Z2v
; 5mm bump and lift
G0 X160 Y197
G0 Y192 Z7 F1000
G0 Y220
G0 Z2v
; 5mm bump and lift
G0 X160 Y197
G0 Y192 Z7 F1000
G0 Y220
G0 Z2v

; centre head ready to clear bed
G0 X170 Y220
G0 Y0 F1000 ;wipe forward
G0 Y280 Z10 F8000 ;wipe aft

dzid26, 2020-05-25:
Just wanted to report that `M190 R30` on **MP Select Mini V1** also doesn't behave right.  Instead of going to set target it controls to whatever the last actual temperature was and stays there. (Even though on printer screen correct target temperature is displayed)
As a result it never reaches a target temperature. 

My solution is to just wait 25min before clearing the bed:
```
M18; disable steppers in between commands to reduce heat from motherboard
M140 S0; set bed temperature for cooldown
G0 Z110 ; lift to the top
G0 X110 Y130; move to back right corner to relax bowden cable
G4 S1500; wait 25min for temp stabilization
G0 X0 Y130; move to back left corner
G0 Z1v; come down to 1MM from bed
G0 X0 Y0 ; wipe forward taller parts (with rail)
G0 X60 Y0; center head
G0 X60 Y110; wipe back lower parts (with extruder)
G28 ; home
```