nukeem/continuousprint #2: Feature Request: Manual Clearing w/ Pause

SrgntBallistic, 2020-04-25:
None of the models I currently want to print lend to having an auto clearing script.

Would it be possible to add an option that starts the next print but immediately pauses before homing?

Those who would like to manually clear their print could do so then hit the knob on the printer to start the print again.

Alternatively I noticed a user working on a script that would possibly take a GPIO input and send the "Start Print" command to OctoPrint which would just reprint the same file.

https://community.octoprint.org/t/repeat-last-printjob-with-physical-push-button/5741

My use case if you're wondering is running 4-6 printers all printing face shield bands, mask ear savers, and pleat folders. Many of these can't be easily auto cleared. Most I have set up to finish all around the same time. So I'd like to be able to walk to my printers, clear each one and then just start the same print or a print from a queue with the press of a button. Without having to go through a bunch of OctoPrint tabs to hit the print button.

Thanks

KlaymoreGhost, 2020-04-25:
You can do this by changing the gcode stop script in the settings window and using M0 (M-zero) at anywhere in the stop script or as the entire stop scrip. M0 is an unconditional stop that waits for the user to click to continue if I remember correctly. I haven't tried it, but so far it 'looks' like the plug-in is just adding your stop script between each job then telling octoprint to run the next job.

SrgntBallistic, 2020-04-25:
> You can do this by changing the gcode stop script in the settings window and using M0 (M-zero) at anywhere in the stop script or as the entire stop scrip. M0 is an unconditional stop that waits for [...]

Thanks for the info. I'll try this and will post back here.

nukeem, 2020-04-26:
>  I haven't tried it, but so far it 'looks' like the plug-in is just adding your stop script between each job then telling octoprint to run the next job.

Exactly right, the bed clearing script in the setting is totally customisation for this reason.

closed 2020-04-26 by nukeem