nukeem/continuousprint #3: Folders are not read

gestadieu, 2020-04-25:
Hi, 
thanks for the plugin, I want to try it for some projects but I already notice that I can't add files in the queue if they are inside a folder, is it a bug or am I missing something?
I was also thinking that maybe adding an icon to add to the queue from the Octoprint list of files/folders could be easier? The icon could be next to those already existing: Download, Remove, Load, Load and Print. 
What do you think?

nukeem, 2020-04-26:
Thanks for the suggestion, that was actually what i attempted in the first place but found it difficult to get the button to work correctly.
As far as i'm aware the files should include folders but i don't tend to use folders so didn't test this. I'll take a look

nukeem, 2020-04-26:
You were correct that folders were not being shown, i have updated to include all gcode files with a search for easier finding.

closed 2020-04-26 by nukeem