nukeem/continuousprint #4: "Complete: example.gcode" message gets shown too early

Andreas-Menzel, 2020-04-25:
When the plugin starts the next print, it displays "Complete: example.gcode" and removes the file from the list. The print however didn't finish yet.
It would be better to mark the print as "currently printing" and show the "Complete: ..." message AFTER the print finished.

nukeem, 2020-04-26:
You may be running an older version of the code somehow as the Complete: message shouldn't be shown.
The plugin uses the octoprint PrintDone event so i have no control over when it notifies me that the print is complete. I have however changed the message to only show starting of new prints or complete queue.

Andreas-Menzel, 2020-04-26:
Okay, I will test it again later. If everything works without a problem, then I will close this issue.

closed 2020-05-06 by Andreas-Menzel