nukeem/continuousprint #6: Multiple of the same file

benbeezy, 2020-04-26:
I'd love to use this to print the same file time and time again. Say I wanted to print a single file X number of times or even an option of looping the same file with no end. 

Right now it lets me add the same file multiple times but when the print finishes it marks all of the iterations of the file as complete.

nukeem, 2020-04-26:
Adding multiple of the same file should work for this, i doubt you would really want indefinite prints as eventually you would run out of filament.

I'll take a look into it, it should just queue the next print regardless of if its the same file or not.

nukeem, 2020-04-26:
I've tested this and it works as expected for me, that's to say if i put the same item in the queue multiple times it will print, clear, print, clear etc the same file.

I have added a new release however so please update and see if it works for you.

radekson, 2020-04-26:
Yes, I would like to continuously print indefinite pieces. . I have MMU2s with spool join function and it is possible to print really long.

MatthewDriver, 2020-04-27:
I would also love an indefinite print option. 
I print a small widget by the thousands. I'd love to print it until the spool is empty. 
I'm so glad to see someone coding this plugin. It the missing link in my continuous printing goal.

nukeem, 2020-04-27:
OK, ill have a think about how i could achieve this

MatthewDriver, 2020-04-27:
A design thought:

A. Infinite printing
Add a checkbox form field "Infinite printing" above the print queue. Once checked only one item can be added to the queue. It will print infinitely (just don't pop it off the queue).

B. Multi qty printing
If 'Infinite printing' is unchecked, when items are added to the print queue, add a 'Qty' form field next to each item on the queue.
Setting a qty of 5, prints five before moving to the next item (and qty) on the queue.
In your python code decrement the qty column before popping the item off.

This should provide for all the flexibility people could ever ask for.

Thanks for considering!