nukeem/continuousprint #8: Can't start queue

taker218, 2020-04-27:
I just installed the plugin on my two octoprint instances, one is running an Ender 3 with Marlin 2.0, the other one is running a CR10 with Klipper.

On the Ender 3, everything works fine (need to finish the first print, but it starts).
On the CR10, I can't start the queue.

In the octoprint.log it says:

> tornado.access - ERROR - 500 GET /plugin/continuousprint/startqueue (CLIENTIP) 114.89ms

(just removed my client IP).

Any ideas why this is happening?
Or is there any other log I could have a look at?

nukeem, 2020-04-28:
Is there any further detail in the log? Unfortunatly this error could mean a lot of things, it does seem odd that it works on one and not the other as apart from the cleaning GCODE (which should be pretty standard) theres nothing thats printer specific.

Can you also check you are using the latest version 1.0.1?

taker218, 2020-04-29:
No more in the log, but strangely yesterday it worked like a charm on both printers, don't know why it wasn't working in the first try.

I restarted octoprint between the tests because I installed another plugin, maybe the power off/power on just happened to help again :D

Will close the issue here.

closed 2020-04-29 by taker218