nukeem/continuousprint #9: Updating to 1.0.1 Fails

SrgntBallistic, 2020-04-27:
Received a prompt to update the plugin to 1.0.1 this morning. I click "Update now" and the installations begins.

[image]

At the end of the installation I notice it says it successfully install 1.0.0. I check the plugin in the manager and it indeed says it's still on 1.0.0

[image]

taker218, 2020-04-27:
Had the same issue.
Just had a look at the setup.py and the plugin_version is still 1.0.0 so I guess you have the newest version, but @nukeem needs to change the version in the setup.py so you don't run into this update loop :)

ViPeR5000, 2020-04-27:
Same here.
Upgrade loop :D

fabien78, 2020-04-28:
It is the same for me.

nukeem, 2020-04-28:
Thanks for letting me know, annoyingly the change didn't push and i didn't notice. Should be ok now.

fabien78, 2020-04-28:
Unfortunatly not.
It is the same issue.
I saw a little thing which, maybe, will help you to understand the problem. At the end of the updating message, it is written:
"successfully installed continuousprint -1.0.0"

I hope you understand me well besouse my english is a little poor....

taker218, 2020-04-29:
Seems like you need to add a new release, or else octoprint will always download the master.zip from the faulty 1.0.1

Maybe just change the version number to 1.0.2 and add a new release, that way it should download the new relase.

nukeem, 2020-04-29:
Sorry everyone this is my first octoprint plugin. Just done a new 1.0.2 release now.

ViPeR5000, 2020-04-29:
same loop again :/

nukeem, 2020-04-29:
I've deleted the previous release, though octoprint is still trying to download it for some reason. I'm wondering if there is a delay before octoprint notices the new release

nukeem, 2020-04-29:
Its worked for me now, required an octoprint restart

taker218, 2020-04-29:
you need to force the update cache to be renewed.
Under settings -> Software update -> Advanced (or whatever it is in english) and then force update

With this it worked for me too. No update loop here on both my instances.

fabien78, 2020-04-29:
Normal process for update is good now.

I don't see differences betwen the old version (but maybe nothing visible for the user)

nukeem, 2020-04-29:
Nothing visiable except a search bar on files list

SrgntBallistic, 2020-04-29:
I was able to use the force update on one of my Pi's to get it to update to 1.0.2. On another I couldn't get it to work with the exact same steps and numerous tries.

I ended up having to uninstall and reinstall the plugin. Obviously backup your gcode first if you have to take these steps. I did the full uninstall where it deletes all plugin data.

I have 2 more Pi's/Printers running prints. I'll see how well they update after they finish.

closed 2020-05-04 by nukeem