nukeem/continuousprint #10: Enhancement idea: Pause between prints

taker218, 2020-04-29:
So, I really like the plugin but it seems like every pause command I use in the script doesn't work.

Would it be possible to make a setting so octoprint pauses between the prints?
I don't know if it is possible, since the print is finished for octoprint and I guess that's why the pause doesn't work as it should.

As a workaround I put in a "G4 S120 ; Dwell for 120 seconds" in the gcode script so I have enough time to remove the print from the bed.

nukeem, 2020-04-29:
You should be able to use a M0 command though i haven't tested this yet. Quite possibly it doesn't work correctly as its not within an actual print. 

I'll keep in mind the idea of adding custom commands in the bed clearing gcode for pausing and resuming the queue manually for the next release

waseltzer, 2020-05-01:
I tried using M0 on my Ender 3.
Didn't work. 
Started printing the next job in the queue without pausing.

nukeem, 2020-05-04:
Yes noticed this too when i checked. i think its because its not actually printing so it get ignored. I've added a [PAUSE]  command in the latest version (1.0.3) which can be used in the bed cleaning gcode to pause indefinitely until you press the resume button.
This only paused the queue however so it will still continue the bed clearing gcode

rjelbert, 2020-05-04:
I have just upgraded to 1.0.3. How can I "press" the resume button electronically? I'd like to do this once I know the bed is clear and the part has dropped into the bin. Thanks.

taker218, 2020-05-04:
Works for me :)

@rjelbert I think you should open another issue for this request.

closed 2020-05-11 by taker218