nukeem/continuousprint #15: wait for GPIO pin before going onto the next item in the queue

rjelbert, 2020-05-02:
It would be very advantageous to have a safety mechanism in this script that waits for the printed part to drop into a collection bin detected by a mechanical switch or beam break switch. Having the code to wait for a low or high GPIO condition would allow someone to add this safety mechanism.

nukeem, 2020-05-05:
I agree this makes sense, and is potentially doable but would limit the plugin to be available on octoprint on raspberry pis.

How it detects would also depend on what type of switch is used NC/NO etc so would need a lot of thinking about. I'll leave this idea open for people to give their 2 cents but this won't be a feature i will implement at the moment unless there is high demand

taker218, 2020-05-05:
Maybe you could implement an API in some sort?
That way it would be more versatile and adaptable to other platforms.

I think the basic API should consist of a status and resume (when you use the [pause] in the script).
That way you could trigger the next print in the queue with whatever you like, be it a switch, a button on a homepage or a button on the printer, which is connected to GPIO and there is a script watching for a change on the GPIO and triggering an API call.

rjelbert, 2020-05-05:
I think there will be a lot of octoprint users with Raspberry Pi so GPIO triggering would be quite an inclusive feature IMO. Other plugins allow you to select which GPIO pin to use and if it is neg or pos triggering. The RPi does have inputs that can have an internal pullup resistor so the simplest input would be active low (to continue). The enclosure plugin is well worth a look at because it offers a lot of flexibility around GPIO.