nukeem/continuousprint #16: Starts without reaching cooldown

clarkent86, 2020-05-04:
While testing the plugin on my Ender 3 Pro the clear job went off well before the bed had reached the default temperature configuration. I'll try testing it again and record more of the values when the clear job went off if it triggers early again. It did seem like the clear job started at 10 minutes but I know there is no timing configuration.

clarkent86, 2020-05-04:
It's also important to know that the clearing failed since the bed was still hot and adhesion was still very strong. Octoprint went to try to start the next print before I cancelled. I would have thought there would be some sort of fail/safe if there was a failure in the clear - this seems like a huge oversight. Maybe it's specific to my printer - but if that's the case compatibility should be called out.

nukeem, 2020-05-05:
We've noticed that on marlin the M109 command doesn't wait for the bed to go cold, see the discussions on #1 where we discuss different options. I myself am doing a wait for 30 minutes.

Not sure what we could do as a fail safe.

closed 2023-12-25 by clarkent86